softerhardware/Hermes-Lite2 #1: Hardware: Wrong voltage for 1.2V supply

softerhardware, 2016-12-30:
Some schematics out in the wild have an incorrect voltage divider (swapped resistor values) for the 1.2V power supply. R14 should be 10K and R15 should be 20K. R14 is the resistor in parallel with C19.

jimahlstrom, 2016-12-31:
On the KiCad schematic in Hermes-Lite2/hardware/hl the value of R14 is 10K, and R15 is 20K.  But in the printed circuit board editor, the values are reversed.  It is necessary to create a new netlist and read it into the board editor.  Then create a new BOM.  I can try this if I have permission.

softerhardware, 2017-01-01:
I have custom Python scripts that generate the BOMs in a different way directly from the schematics. I just need to update all files eventually.

jimahlstrom, 2017-01-01:
There are a number of small PCB changes (labels, etc.) in these Open Issues.  Do you want me to make them and push them to github, or do you want to make them yourself?

softerhardware, 2017-01-02:
I am happy if you make these changes. I do not want give commit privileges to others yet for HL2. (HL1 and HL2 are separate repositories.) It is under active development, and just like what is done with Linux, I want to review all accepted patches. Please create a fork and then issue a pull request for the changes you think should be in my repository. See https://help.github.com/articles/creating-a-pull-request-from-a-fork/ Please make more smaller commits to your github rather than fewer large commits as that will make it easier to pull in changes. I will try to do the same to make merges easier for you too.

In my mind, a successful open source project accept patches from others, but has a gatekeeper that reviews patches, even when they come from skilled and qualified contributors. I've seen too many open source projects devolve into a buggy bowel of spaghetti because everyone is given commit privileges.

Sometimes a fork overtakes the original repository because the fork developer is more active, does better work, or has better ideas and goals. I am totally fine with that if it happens.

jimahlstrom, 2017-01-02:
Perfectly fine with me.  I will fork and issue a pull request.  My git is a little weak, so if I am doing it wrong, please advise.  Meanwhile just assume I am working on all PCB issues.

jimahlstrom, 2017-01-02:
Posted here in case someone else wants to work on the PCB.

Without any PCB changes, I made a new hermeslite.net file on my Linux KiCad version 4.0.2.  Your version was made on Windows KiCad 4.0.4.  The files seem totally different because of the difference in line endings between Windows and Linux.

So I made a new hermeslite.net on Windows KiCad 4.0.5.  Now the line endings agree but the new version wants to update the obsolete library files.  I could use 4.0.5 but I am not sure your 4.0.4 will read the new files.

To make things simple, anyone working on the PCB must be using the same operating system and the same version of KiCad.  I don't have your version of KiCad 4.0.4 and I don't know where to download it, as the new 4.0.5 is on the download page. 

I am happy to switch to Windows, but either you must upgrade to 4.0.5 or I must find the install file for 4.0.4.  That way, there should be no confusing differences in the files, and the only difference should be the result of actual changes.  And it will be easy to compare the original and changed branches.

softerhardware, 2017-01-03:
I switch back and forth between using 4.05 on Linux and 4.04 on Windows without issues. The KiCAD databases/files are compatible when just the minor version number changes. KiCAD can handle the files with or without the Windows control-M at the end of lines. There are also the Linux utilities dos2unix and unix2dos to clean this up. These utilities are also available with Cygwin when using Windows. I will upgrade my Windows install to 4.0.5. You can find the Windows install at http://kicad-pcb.org/download/windows/

jimahlstrom, 2017-01-03:
Glad to hear you are set up for dual boot, and able to sort out line end problems.  Another handy command with Ubuntu is:
git difftool --tool gvimdiff origin/master rtl/receiver.v
The gvimdiff tool is immune to line ends, and you can change the last two file arguments or omit them.

I will carry on on Linux with KiCad 4.0.2 if that is OK with you.

softerhardware, 2017-01-10:
I had switched the values of R14 and R15, but didn't generate a netlist. You switched them back. I just switched them again and updated the netlist. Also, in the schematics, I don't like to use the hermeslite-cache library as that is problematic, so removed it.

closed 2017-01-10 by softerhardware

jimahlstrom, 2017-01-10:
The schematic and PCB have correct values for R14 and R15.  But the BOM is incorrect.  I tried to run your code to create the BOM but it failed because some imports were missing.  Please run the BOM code, as it depends on the netlist file.

softerhardware, 2017-01-11:
I updated the BOM pdfs. If you like, you can grab missing packages with "pip install <package name>." LaTeX is also required to create the pdf.