softerhardware/Hermes-Lite2 #10: Hardware: FPGA EEPROM on some BOMs is too wide

softerhardware, 2016-12-30:
The FPGA EEPROM U1 on some BOMs is specified as the wide package. The footprints on the PCB expects 3.9mm width. All BOMs in circulation should be updated.

jimahlstrom, 2017-01-10:
The part number on the BOM is incorrect.  It should be S25FL116K0XMFI041.  In the last four characters, the "I" (eye) is the temperature grade, "04" is the narrow 3.9 mm width, and "1" means tube packaging.  Mouser doesn't stock this, but they have the better temperature "A" part with Mouser number 727-S25FL116K0XMF040 for $1.36.  Digikey stocks S25FL116K0XMFI041 with Digikey part number 1274-1106-ND for $0.53.

I do not know how your database BOM files work, so I can't change this myself.

softerhardware, 2017-01-11:
Please add any alternate new part numbers to Hermes-Lite2/hardware/hl/bom/parts.json. This is a text file. If you look for "C 270pF 0805" in this file, you will see that there is a list "mpn" associated with it of part numbers. You can just add new part numbers to this list. The requirement is that the part number yields a valid result on www.octopart.com. Once the new parts are in place, they will be picked up and possibly used the next time I generate a BOM.

jimahlstrom, 2017-01-11:
I didn't change anything, but the "I041" part number is now correct in parts.json.

softerhardware, 2017-05-25:
Corrected.

closed 2017-05-25 by softerhardware