softerhardware/Hermes-Lite2 #11: Hardware: Oscillator X2 footprint is very tight

softerhardware, 2016-12-30:
The primary X2 on the BOM has 6 pins. Only the 4 outer pins are used and the 2 inner pins are no connect. The footprint is such that it is easy to short one of the outer pins with an inner pin. Adjust the footprint to make this less likely. Also, the primary X2 has no pads that wrap around the edge for easy hand soldering. I resorted to hot air for this (which worked well). The current X2 is preferred as it is the most stable. Consider specifying/recommending a 4-pin variant also on the BOM for any hand builds.

jimahlstrom, 2017-01-05:
I am working on this.

jimahlstrom, 2017-01-09:
The footprint for X2 supports multiple packages.  A useful trick is to place all the manufactures' recommended footprints for various packages on the bottom, and then draw a footprint on the top that accommodates them all.  When the top is done, delete the bottom footprints.

There doesn't seem to be a way to specify a keepout on a footprint, so KiCad will try to flow ground copper under X2.  You must draw the keepout on the PCB every time the X2 footprint is used.

closed 2017-01-10 by softerhardware