softerhardware/Hermes-Lite2 #12: Hardware: Add alternates to BOM

softerhardware, 2016-12-30:
The current BOM only includes the preferred part for any line item. There are more alternatives in the master database. These should be exposed in the BOM so that builders can better see and understand the options.

jimahlstrom, 2017-04-05:
The commercial receive transformer T2 is:
    MABA-010143-FLUX18
The commercial transmit transformers for T1 are:
    CoilCraft PWB1010-1LB
    MABAES0060 from Mouser
    ETC1-1T 
    Mini Circuits TC1-1TG2+
The above parts were discussed on newsgroup "Commercial Transformers" on March 14, 2017.

daniestevez, 2017-04-11:
Low priority: The EEPROM part for U1 (S25FL116K0XMFI041) in the BOM should be replaced by an equivalent part. I just received notification from Digikey that this particular part reaches its end of life.

softerhardware, 2017-06-20:
I considered this but don't have room in the pdf. I or other users can always generate a new BOM with alternatives using the scripts.

closed 2017-06-20 by softerhardware