softerhardware/Hermes-Lite2 #17: Firmware: Sync HL2 Ethernet RTL with latest openHPSDR version

softerhardware, 2016-12-30:
The HL2 ethernet RTL forked from the openHPSDR version over a year ago. It is now quite different to support the old protocol and we can't just switch to what openHPSDR has. All changes made to the openHPSDR since we forked should be reviewed and incorporated into the HL2 version if of value.

jimahlstrom, 2017-01-01:
Does anyone have a URL for Phil's current Ethernet code?  Does this code use the same MAC as HL2?

softerhardware, 2017-01-02:
I always have to ask Phil for the latest RTL. Last time I asked, I didn't get around to merging it into our branch. Before asking him again, I want to be sure that I am ready and will merge his changes soon.

I forked from Phil's new ethernet MAC more than a year ago. The new MAC does not support the old protocol. The changes we made to this MAC was to support the old protocol and fix a few bugs. The HL2 MAC and Phil's new MAC are now different enough that you can not take the latest from Phil and plug it into any HL RTL and have it work. Phil may have fixed some other bugs and added support for 100 Mb/s since we forked which I would like to merge back into the HL2 repository.

jimahlstrom, 2017-01-09:
I got a new ./Ethernet directory from Phil because we are working on the DHCP renewal.  We have new DHCP code in the files Ethernet/dhcp.v and Ethernet/network.v.  These two files are now in Hermes-Lite/rtl/Ethernet and (I think) can be directly moved to HL2.

softerhardware, 2017-01-10:
Great! Can you e-mail me a copy?

softerhardware, 2017-05-25:
Check done in early Jan. Not much difference. Now openhpsdr supports 100mb so another check should be done.

softerhardware, 2018-02-18:
Check done in January 2018. Incorporated some DHCP enhancements from Jim. With 1000/100 autosense, we are now quite different and more enhanced then the main branch.

closed 2018-02-18 by softerhardware