softerhardware/Hermes-Lite2 #18: Firmware: Support for i2c programming of clock and PA bias

softerhardware, 2016-12-30:
As a first step, create a bit file that properly programs the i2c clock generator.

As a longterm solution, add support for a wishbone communication structure to all i2c and spi clients so that they can be accessed via software over a ethernet connection sidechannel.

jimahlstrom, 2017-01-01:
I looked up "Wishbone" on wikipedia.  Interesting.  But it is easier to extend the Hermes PC to HPSDR protocol by using the C0 indexes from 18 to 127 that are currently unused.  For example, we need a way to specify the frequency of the second clock output of the VersaClock.

softerhardware, 2017-01-02:
Wishbone is part of www.opencores.org. I work with a lot of RTL in may day job, and there are certain design patterns that lead to RTL that is maintainable, sharable, learnable, less buggy, etc. One of those is abstraction of IP blocks and a standard mechanism (bus) for communication. This is lacking in the current inherited hermes-lite RTL. A wishbone bus provides this.

I do not want to simply extend the protocol because most people are not using software that natively supports the Hermes-Lite like Quisk or Spark SDR. They are using PowerSDR. By adding a side channel to make Hermes-Lite features, people can continue using PowerSDR with support for Pure Signal. On a side note, Quisk could use Warren Pratt's wdsp library and implement pure signal too.

My goal is to have everything memory mapped. For example, to specify the second clock output of the VersaClock, the software will read and write memory locations that correspond to a remapped range of the VersaClock IC's native memory map. The software will program with the native VersaClock interface. There will be no abstraction in the RTL that takes a frequency and converts this to VersaClock interface commands. This leads to simpler and smaller RTL, and more flexibility for the software.

I have thought about and am open to extending the current Hermes protocoal (via C0 indices fro 18 to 127, etc.) to pack memory map read and write commands into the existing protocol. This would allow software targeting the Hermes-Lite to have just one communication mechanism. The RTL would support both paths, side channel and packed in original protocol, to support PowerSDR and other software users.

jimahlstrom, 2017-01-02:
All this sounds good to me.

As you know, it is hard to make a "universal protocol" for hardware you don't have yet.  The current Old Hermes protocol is full of peculiar bits of data for old hardware which will probably never be used again.  But it lacks anything like the 2nd output of VersaClock.  This is a tough problem.

softerhardware, 2017-05-25:
Wishbone bus in use for AD9866 and I2C peripherals. Clock, bias and slow ADC i2c in place. Need to combine clock and bias I2C controllers to save space in the FPGA.

softerhardware, 2018-05-19:
Switched to cmd bus with ack and AXI4-ST for stream traffic.

closed 2018-05-19 by softerhardware