softerhardware/Hermes-Lite2 #5: Hardware: Simplify metric hole sizes

softerhardware, 2016-12-30:
In beta2, corner holes support metric M2.5 but the hole for pressure on the PA transistors is M2. Consider increasing this hole to M2.5 to simplify nut/bolt/standoff requirements.

vk3pe, 2016-12-30:
Steve, not sure about USA but in VK, 2 or 2.5mm screws are much harder to get in small quantities, than 3mm. (even though we are an official metric country)  Might you consider changing all to 3mm?
glenn
vk3pe

vk3pe, 2016-12-30:
Steve, from Stews picture,  it appears the holes on the rear of some parts for soldering the IC paddles, is located in a corner of the chip. Would it not be better located in the centre?  And made a bit larger? I find 3mm a better size for getting a reasonable sized solder iron tip through for better heat transfer. (ie a small hole means a small solder iron tip which is much harder to get good heat transfer, to solder the chips paddle.)

glenn
vk3pe

vk3pe, 2017-01-01:
> Glenn, there was not enough room for the screw/bolt heads of 3mm
> hardware the last time I checked. I will check again. The smaller
> sizes are available from Digikey and Mouser. My local hardware store
> carries some imperial small sizes that work fine too. We should
> document what common hardware will work.
>
> The holes are off center on purpose. Usually the IC die is in the
> center and I want the best heat transfer from the IC die to PCB. The
> primary goal is good heat transfer for automated assembly of boards.
> The hole is just there to facilitate manual assembly for a minority of
> builders.

Thanks Steve

You need about 240-250mils for a 3mm screw head.  Digikey and Mouser is 
fine but freight cost are the killer for us. OK if already placing an 
order for other items though.   Most electronic stores here (we have 
only have 3 companies with outlets) stock 3mm screws, sometimes 2.5mm in 
very limited lengths and never 2mm.. Imperial sizes just not available 
in small qty. I am talking local stores, not via internet. I don't have 
a pcb in my hand and you know it better of course, but the step up from 
2.5mm to 3mm is not all that great.

I would have thought that auto assembly of PCB's with a hole off centre 
of the paddle, would tend to draw heat away from the opposite corner? I 
may well be wrong but center 'seems' like a nicer way. And 3mm even 
better to fit a decent solder tip in the hole for good heat transfer, 
not just a blob in the hole, which may not solder the paddle at all. 
Just my thoughts.

glenn

vk3pe

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

ahopper, 2017-01-01:
Steve,
I'd vote for M3 if you can make it fit, taps get very fragile below M3 and tend to be more expensive.  The average cheap tap and die sets often stop at M3.
Alan M0NNB

softerhardware, 2017-01-02:
My measurements have been based on http://www.spaenaur.com/pdf/sectionR/R11.pdf My concern is not the hole size for M3 but the head size of the bolt or screw. Looking at the PCB, we should be able to support a M3 head size of 5.6mm. I will have to expand the hole to accommodate M3. Note that not all M3 hardware will work, as some head dimensions for M3 are 6mm.

Since most diy builders will have to order some specialized parts (AD9866, ethernet, et.c) from a place like Digi-Key or Mouser, I don't think it is too much to expect that some hardware should be added to the order. Also, the mounting holes are secondary in my mind. An enclosure is needed for proper thermal dissipation of the PA and so fitting into the slot of an enclosure is more important in my mind.

I've already received concern from assembly houses regarding the hole I include to solder the hidden pad on some ICs. This is nonstandard and only included to support hand builders. Again, I expect the number of hand builders to be low relative to those who want a preassembled unit so this hole is a secondary consideration. Solder is a relatively bad conductor of heat. You want the center of the pad as close as possible to copper and vias for thermal dissipation. The empty space with a centered hole means the solder is conducting the heat. Since Stew has been able to build his board with an iron, I feel the holes are large enough.

vk3pe, 2017-01-02:
Thats all fine Steve.

Happy to go with whatever you decide.

glenn

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

ahopper, 2017-01-02:
just to complete your list of head sizes http://www.roymech.co.uk/Useful_Tables/Screws/cap_screws.htm this puts M3 socket head cap screws at 5.2-5.5mm
Alan

softerhardware, 2017-06-20:
PCB supports M3 screws with small heads, regular M2.5 in all corners plus the pressure point for thermal dissipation. 

All other holes on PCB have been checked and consolidated.

closed 2017-06-20 by softerhardware