softerhardware/Hermes-Lite2 #7: Hardware: Working support for multiple enclosure widths

softerhardware, 2016-12-30:
Beta2 has tabs to support two enclosure widths. The intent is that these tabs may be broken off for use with  the narrower enclosure width. The beta2 boards were fabricated without V-cuts to make this break-off easy. Also, the enclosure listed in the BOM has a slot that is not tall enough for a 1.6mm thick PCB. Support for multiple enclosure widths needs to be cleaned up. Possibilities are:

Work with Elecrow so that V-cuts are on the next batches of boards, and use a 1mm thick PCB (no extra cost) instead of 1.6mm. This option is attractive as PCB weight will be reduced for shipment and 0.05 spacing inexpensive headers/receptacles can be used for end launch connection to the filter board.

Scrap the narrow enclosure option, but then the BOM must be updated with a 10mm length of the current 15mm length enclosure. The 15mm length is commonly available, but the 10mm length is supposedly available for special order. I prefer relying on commonly available enclosures.

jimahlstrom, 2016-12-31:
I recall that the enclosure first discussed by Steve had a slot for the PCB, but the eBay reference for the enclosure EN1 lacks a slot; or at least the picture lacks a slot.  It is also out of stock.  This enclosure

http://www.ebay.com/itm/Silver-Aluminum-PCB-instrument-Box-Enclosure-DIY-Project-100-100-50mm-US-Stock/112121507828?_trksid=p2047675.c100623.m-1&_trkparms=aid%3D222007%26algo%3DSIC.MBE%26ao%3D1%26asc%3D38530%26meid%3De6752ed3afa14a93b7b78d6fd7172592%26pid%3D100623%26rk%3D2%26rkt%3D6%26sd%3D322239636151

is available but I can't quite make out if it has a slot.  Even if it does, I don't know the size.  The two comments state that the size 100x100x50mm is inaccurate.  I am uncomfortable depending on boxes on ebay that come and go and vary in size.

In contrast, the Hammond 1455 series boxes are designed for standard PCBs and have 2mm slots.  For example, the 1455L1201 fits a 100x120mm PCB.  But it costs $19.32 from Digikey.  It has multiple PCB slots, so is convenient for inserting a second board.  But the slots interfere with mounting a TO-220 on the side of the box with good thermal contact.

I am uncomfortable with a 1mm PCB  because OshPark does not offer them.  I guess if we just provide everyone with boards from Elecrow maybe the non-standard size is not a problem.  But why change to a 1mm board because of an ebay board that goes out of stock.

Maybe going to a no-slot box makes sense.  The HL2 is mounted with four standoffs, and more boards are added with a second layer of standoffs.  Then the box is less of a problem.

I guess I came up with more problems than solutions.  Sorry.  I will try and search for more boxes to see if there is a standard size we can depend on.  If anyone knows of one, please post here.

softerhardware, 2017-01-01:
Jim, I have both target enclosures in hand. They are pretty common models and I don't anticipate variation. I have verified the slot is not tall enough in one unit. I also have drawings for these two enclosures. I want to stick with the two current target enclosures, although hopefully keeping it easy to target other enclosures for those who want to.

I don't think a 1mm PCB will be a problem for you or any others interested in using OshPark. The same Gerber files are used for both 1.6mm or 1mm. You can have OshPark build 1.6mm. When ordering from www.elecrow.com, one just select the thickness they want. People using OshPark may want to tweak the final card dimensions also to support some other target enclosure they want to use. 

I don't think the stackup spacing (except between the two inner layers) or dielectric variations between OshPark and Elecrow will be significant enough to matter. A closer spacing between the two inner ground layers should have no electrical impact. The 1mm thick boards have better thermal transfer for the PA transistors. The PA test boards sent to Claudio were 1mm thick.

jimahlstrom, 2017-01-01:
There is one enclosure on the HL2 BOM.  What is the URL for the second enclosure?  If the dimensions are standard then we can provide a template for drilling the holes in the back panel.

softerhardware, 2017-01-02:
The second enclosure is:
https://www.amazon.com/Electronic-Enclousure-Amplifier-Headphone-Wv-diyc-l/dp/B00ZOM3BUK/ref=sr_1_7?ie=UTF8&qid=1483327305&sr=8-7&keywords=aluminum+electronic+box+diy

Search for AK-C-C12 on www.aliexpress.com. SMOZK will customize the length. 150mm is the target with filter extension board, 100mm without the extension board. There are other manufacturers of this style of enclosure. Search for aluminum enclosures 150*105*55  or 150*106*55 on aliexpress or google.

Another possibility for the panes is a 3D printer. I expect that elecrow will be able to help with sourcing enclosures with custom front and rear panels once we go into small production.

softerhardware, 2017-06-12:
Since it appears difficult to have PCBs fabricated with break away tabs for various widths, beta3 will be 100mm. There is unused area so that narrower width PCBs can be fabricated in the future if needed.

For enclosure, I am standardizing on AK-C-C12 my SZOMK and others at 100 mm length for two board configuration, filters on second 100x100 board. Google AK-C-C12.

closed 2017-06-12 by softerhardware