softerhardware/Hermes-Lite2 #22: Hardware: make sure stuff is disabled during FPGA configuration

in3otd, 2017-02-22:
In the H-L v2b2 the TX relay toggles at power-up, while the FPGA is being configured. Similarly, the  bias source for the PA, U14, is briefly enabled. May not be harmful but would be good to add a pull down (as R125, which should likely be mounted by default).
I find the relay clicking a bit annoying because it makes the user think that the unit went into TX unexpectedly.

in3otd, 2017-05-22:
I may have missed something but why did you move this to "Firmware"? My understanding was that you cannot control the pin status during configuration - the docs say "The user I/O pins and dual-purpose I/O pins have weak pull-up resistors, which are always enabled (after POR) before and during configuration".

softerhardware, 2017-05-23:
Yes, you are right. I didn't pay close attention when reading this initially. There are also some sequencing changes that need to be made in the firmware.

softerhardware, 2017-05-25:
Resistors R124, R125, R126 added to beta3 schematic.

softerhardware, 2017-06-10:
Done on PCB.

closed 2017-06-10 by softerhardware