softerhardware/Hermes-Lite2 #23: Hardware: Feedback from Claudio

softerhardware, 2017-03-04:
General:
- could be useful to add to the wiki that KiCad 5 (under development) should allow to flip the PCB view; this will make easier to find/check components on the bottom side. This new feature was mentioned at FOSDEM 2017, I did not try KiCad 5 yet.
- for those manually assembling an H-L v2 from a bare PCB, a separate top-components BOM and bottom-components BOM will be useful. I extracted the top/bottom info from the KiCad "PCB Footprint Report" with a simple python script and manually marked with a different color every component on the BOM. Maybe your scripts could be modified to print the references in the BOM e.g. in red for the components on the top and in green for the ones on the bottom so there will still be a single BOM but it will be immediately clear where a component goes. This can avoid flipping the PCB too many times or easing placing all the components on one side first for those using reflow.

Schematic:
- U12 should be marked DNI ?
- is there a reason to have connection dots on the lines to DB2, DB16 and DB4?

BOM:
- J29 has a space in front of the reference text in the BOM (just wondering why)

PCB:
- C12 reference positioning on silkscreen unclear (maybe there was no space for a connecting line)
- D8-D11 polarity on PCB unclear (I see no polarity indication in their silkscreen)
- R54 refdes partially over pad
- B70 refdes partially over via
- B22 1v2 via maybe too close to B29, for manual assembly
  The problem here is that vias are not tented, which is great for a prototype, since one has many test points for "free" but for (manual) reflow may lead to some issues.
  On my PCB, after reflow I found several capacitors which were "skewed" toward the untented via, probably because of solder sucked by the via and/or surface tension during reflow moving the component in that direction. I then used too much solder paste and had several shorts due to this. I understand it may not be an issue for a proper reflow process but I wonder if in a "production" run via near pads (or even in general) should be tented.
  Same as above for
  - B10, C3 and B4
  - B9, B3 and B14
  - B18 and B30 - but they are on same net, GND (I just got the caps stuck together during soldering)
  - B19 and B33 - but they are on same net, GND
  - B11 and B16 - but they are on same net, GND
- C48, C150 could be in 0603 (like the other 10 uF caps) instead of 0805?
- X1 has no pin 1 dot/indication on silkscreen
- D1 should be CDSOD323-T05L*C* (bidirectional); its footprint on PCB is fine for an unidirectional device but will not for a bidirectional one (it shows a polarity line, bidirectional devices do not have a polarity line on case)
- L33, L34 (actually 0 ohm for a "standard build") can be 0603 like the other 0 ohm resistors in the BOM

Testing:
- on the wiki, for the leds.jic test, which 2 of the 4 are lit ? (is it random ?)

in3otd, 2017-03-25:
- PCB mounting holes: the holes in the corners currently accept a #<i></i>4 screw or a M2.5. Here standoffs are slightly more common with M3 screws, in my experience - M3 can be forced in the current holes but it's of course not a good thing to do. M3 will require also a bit more clearance, to be safe.
No real need to enlarge the holes, but put a note somewhere to use M2.5 or #4 screws :grin:

- T3 (PA output) holes are too big, except maybe the one for the center tap, which will usually be done by twisting the wire. If you would like to support wires in parallel for the primary, two smaller holes side by side can be used.

softerhardware, 2017-05-25:
Schematic changes made in beta3.

softerhardware, 2017-06-20:
Still need to clean-up silkscreen, but a few comments:

C48 is now uses 1206 package (as others in the power supply do) for higher ESR. There are only two sizes of 10uF now.

L3 and L4 are 0 ohm now.

Mounting holes are larger to supper M3

softerhardware, 2017-06-28:
Vias can be selected now to have tenting on top, bottom or both. 
I think all issues are addressed. It would still be nice to eventually have printouts of components by side.

closed 2017-06-28 by softerhardware