softerhardware/Hermes-Lite2 #29: Test Q5 works, use same device for Q2 and Q5

softerhardware, 2017-03-25:
Test that external PTT driven by Q5 works. 

There are two similar devices Q2 NUD3124 and Q5 2N7002K. I would like to merge these BOM lines. The 2N7002K is less expensive. Can it be used for Q2?

in3otd, 2017-03-26:
I think it can be used, with a freewheeling diode preferably.

We could probably get rid of Q5 entirely and use the unused pole of K2 instead.

kf5kog, 2017-03-27:
Q5 works fine with John's 5W amp.   Using K2 unused pin is an interesting idea.  The SMA connectors can be slid up and the PTT out can be placed near the relay.   Stew

daniestevez, 2017-03-27:
Q5 also works fine with my Hardrock-50 HF amp.

in3otd, 2017-03-27:
out of curiosity I did some measurements around the PA relay timings, using Quisk 4.1.3.hl2; I put a pull up on the PTT output at CN11 and used the relay second pole to switch a voltage and get an indication of when the relay had actually switched (I hadn't soldered these relay pins on the PCB but bent them outwards for later use).

At TX start - with infinite persistence
[image]
the relay switches in a couple of milliseconds; actual RF output starts 30 ms to 50 ms later - seems a bit long even if we are not aiming for QSK, hi. I do not know if this delay is introduced by Quisk or the FW on purpose.

At TX end - also with infinite persistence
(note the change of scale w.r.t. the previous capture)
[image]
The RF output stops just before the PTT output is released, relay switches about 1 ms later.

I was wondering if we need a *TX inhibit* input also; it could be useful for relay sequencing for those using old big relays or transverters. This will exclude the possibility of using the relay second pole as H-Lv2 PTT output, though.

in3otd, 2017-03-28:
> I was wondering if we need a TX inhibit input also; it could be useful for relay sequencing for those using old big relays or transverters. This will exclude the possibility of using the relay second pole as H-Lv2 PTT output, though.

correction: the relay can be used anyway, it's enough that no RF signal is present at the H-L PA output until the *TX inhibit* is deasserted.

softerhardware, 2017-06-20:
NUD3124 is used in both places. Footprint is the same if we need to revert back. TX inhibit is specified to a free input.

closed 2017-06-20 by softerhardware