softerhardware/Hermes-Lite2 #47: Hardware: Switch to 2 input opamp, direct connection for temperature and current

softerhardware, 2017-09-01:
https://github.com/softerhardware/Hermes-Lite2/wiki/Build-Notes-2.0beta3#remove-opamp-from-current-and-temperature-sense-paths

softerhardware, 2017-09-01:
Current measurement is not full range, use resistor divider at input of slow ADC.

Taka has noise or pegged at zero issues. Review his modifications.

in3otd, 2017-09-02:
The opamp used may not be well suited to the diode linearization circuit, as its bias current is relatively high. Moreover, the conventional wisdom says that the RF detector diode and the linearization diode used around the opamp should be matched; this won't likely be the case if these two diodes resides on two different boards. We could maybe drop the opamp completely and assume that it will be on the filter/SWR bridge board or provide jumpers to access the ADC inputs directly. More details in the following days, hopefully  :grin:

softerhardware, 2017-09-24:
U18 was essentially eliminated in beta4 with bypass vias provided for FWD and REV, and temperature and bias current connected directly to the ADC. The issue remains open as U18 will be entirely removed in beta5. This op amp belongs near the SWR bridge.

softerhardware, 2017-10-27:
Removed in beta5.

closed 2017-10-27 by softerhardware