softerhardware/Hermes-Lite2 #72: Bidirectional TVS symbol is misleading

in3otd, 2017-10-27:
The symbol for the SMBJ18CA (D12) and  the CDSOD323-T05LC  (D1) are a bit misleading, they should represent two diodes in "antiseries", like [here](http://www.littelfuse.com/~/media/electronics/datasheets/tvs_diodes/littelfuse_tvs_diode_smbj_datasheet.pdf.pdf).
BTW, I'm not sure that D12 should be a bidirectional diode, even if negative voltages will be blocked by Q1, maybe we could use an unidirectional protection there.

softerhardware, 2017-11-05:
http://datasheet.octopart.com/CDSOD323-T05LC-Bourns-datasheet-13385636.pdf is slightly different.

in3otd, 2017-11-05:
yes and maybe closer to the actual internal design but the complete circuit does not fit well in a schematic symbol :grin: . My point was that the current symbol looks like two Schottky diodes in antiparallel so at a first glance it appears to conduct at around &pm;400 mV

softerhardware, 2017-11-06:
Schematic symbols updated to match datasheets.

closed 2017-11-06 by softerhardware