softerhardware/Hermes-Lite2 #118: Gateware: DHCP Discover Dropped if no Checksum

softerhardware, 2019-07-14:
Some DHCP servers will drop the DHCP discover if the checksum is set to all zeros, not used. This is an issue with some DHCP servers as an all zeros checksum (checksum not used) is a legal alternative.  This issue was created to track this known problem, and possibly add checksum to all UDP packets sent from the HL2 (costly in terms of FPGA resources) or to switch to an ethernet stack that applies checksum to all UDP packets, such as https://github.com/alexforencich/verilog-ethernet

csylvain, 2019-07-16:
what is happening is an intermediate network device is messing up the checksum computation when bridging a DHCP packet with a 0x0000 checksum. in turn, the DHCP server rejects the incoming packet because the bridge has added a checksum and it is incorrect. the DHCP server (ISC 4.1 and later) is capable of replying to an incoming udp packet with no checksum, it just rejects packets with invalid checksums.
i see there is already a crc function in the gateware related to the network functions but did not dig in enough to determine where it is already used.

softerhardware, 2019-12-14:
I'm considering this a problem with DHCP servers and will not support workarounds in the HL2.

closed 2019-12-14 by softerhardware