softerhardware/Hermes-Lite2 #139: Failure to exit transmit mode

wb2qll, 2020-01-07:
When the PTT line from my HL2 build 9 is brought low, the HL2 and PowerSDR goes into transmit mode and the MOX button is lit. This is correct. However, in all modes except for cw, when the PTT line is opened (or the PTT cable removed from the jack) the HL2 remains in transmit mode and PowerSDR remains in transmit mode with the MOX button lit as well. This happens even if VAC is disabled throughout the PTT operation. Happens into a dummy load regardless of drive setting. This is on a Windows 7/32 bit system with the HL2 plugged directly into the computer. Transmit mode on both the HL2 and PowerSDR can be exited by pressing the MOX button.

softerhardware, 2020-01-07:
Which gateware version are you using? Software reports it as 65 or 66, etc., usually at the top of window for Quisk. Can you update to the latest testing and see if that fixes the issue? It sounds very much like a bug that was recently fixed. 

[latest testing gateware](https://github.com/softerhardware/Hermes-Lite2/tree/master/gateware/bitfiles/testing/20191110_68p4)

[gateware update how to](https://github.com/softerhardware/Hermes-Lite2/wiki/Updating-Gateware)

The ethernet method with SparkSDR or Quisk is recommended.

wb2qll, 2020-01-07:
Hi Steve, thanks for the response.I can't get Quisk to work. Python and it installed without errors but when I type "quisk" it just returns the command prompt, no error indication and no processes started. I don't expect you to take time on my issue, but I've never used Python or Quisk before so it may take some time to suss out.PowerSDR reports the firmware as 6.7I do have a USB blaster, which I've never used, but I did see you have a video on running it, so thanks again.Pete-----Original Message-----
From: Steve Haynal 
Sent: Jan 7, 2020 12:27 PM
To: softerhardware/Hermes-Lite2 
Cc: wb2qll , Author 
Subject: Re: [softerhardware/Hermes-Lite2] Failure to exit transmit mode (#139)

Which gateware version are you using? Software reports it as 65 or 66, etc., usually at the top of window for Quisk. Can you update to the latest testing and see if that fixes the issue? It sounds very much like a bug that was recently fixed.
latest testing gateware
gateware update how to
The ethernet method with SparkSDR or Quisk is recommended.

softerhardware, 2020-01-07:
Hi,

SparkSDR will do network gateware upgrade. It is easy to install on Windows. Just right click on the detected radio shown in the top bar main window and select the .rbf file. SparkSDR is found here:
http://www.ihopper.org/radio/previews.htm

I think you are seeing an issue fixed in the 68 release. See the latest testing gateware link sent before and the bug reported by WN7T.

73,

Steve
kf7o

wb2qll, 2020-01-07:
Yes, Steve, thank you very much.I got Quisk working. Jim's web page says, following the install steps "You should then be able to start Quisk with the command "quisk" or "your_install_directory\Scripts\quisk"." I guess that's not right, because that's what I did before. When I typed "python -m quisk" on a hunch it plays nicely!I can follow instructions, but if they're wrong, well....Anyway, it's gateware 67. Yes, it sounds like the same issue WN7T reported, so I guess I'm off to check out Spark.73,PeteWB2QLL-----Original Message-----
From: Steve Haynal 
Sent: Jan 7, 2020 4:12 PM
To: softerhardware/Hermes-Lite2 
Cc: wb2qll , Author 
Subject: Re: [softerhardware/Hermes-Lite2] Failure to exit transmit mode (#139)

Hi,
SparkSDR will do network gateware upgrade. It is easy to install on Windows. Just right click on the detected radio shown in the top bar main window and select the .rbf file. SparkSDR is found here:
http://www.ihopper.org/radio/previews.htm
I think you are seeing an issue fixed in the 68 release. See the latest testing gateware link sent before and the bug reported by WN7T.
73,
Steve
kf7o

wb2qll, 2020-01-08:
Hi again. I followed your instructions and installed Spark and selected the  <https://github.com/softerhardware/Hermes-Lite2/blob/master/gateware/bitfiles/testing/20191110_68p4/hl2b2_main/hl2b2_main.rbf> hl2b2_main.rbf from the v68 github area. Spark ran and said programming was completed.

 

However now neither Spark nor any other software recognizes the HL2. I tried power cycling to no avail.

 

What to do now?

 

Tnx

Pete

 

http://www.ihopper.org/radio/previews.htm

wb2qll, 2020-01-08:
Hi Steve:

 

I really hate taking your time or seeming to bug you, but I have a feeling you’re going to tell me I have to use the USB blaster to get this HL2 working, since Windows software doesn’t find it.

 

I can’t use your nice video, because there is no longer an Altera website and the current files are named in a way that doesn’t resemble anything that you’ve specified. 

 

Unless this is written up someplace else I haven’t found, I’ll need to know exactly what to download from github and from:

 

https://fpgasoftware.intel.com/13.1/?edition=web <https://fpgasoftware.intel.com/13.1/?edition=web&platform=windows> &platform=windows

 

I think. There’s no such entry as “programming files”.

 

I do intend to write up a “getting started” piece once I get a stable system.

 

Thanks again,

Pete

ahopper, 2020-01-08:
@wb2qll Pete, I suspect the programming issue is caused by using the wrong rbf file,  for a build 9 you needed the rbf with 5up in the name. Unfortunately there is no way for the programmer to detect and prevent this at the moment. I think this https://fpgasoftware.intel.com/?edition=lite&elq_cid=2980428&erpm_id=2702282 should have the fpga programmer.

wb2qll, 2020-01-08:
I agree, but there wasn't any way I could tell what file I needed.That site does have the fpga programmer, but since the file names don't match your video. I have NO IDEA what to download out of that .Thanks.Pete-----Original Message-----
From: ahopper 
Sent: Jan 8, 2020 12:45 AM
To: softerhardware/Hermes-Lite2 
Cc: wb2qll , Mention 
Subject: Re: [softerhardware/Hermes-Lite2] Failure to exit transmit mode (#139)

@wb2qll Pete, I suspect the programming issue is caused by using the wrong rbf file,  for a build 9 you needed the rbf with 5up in the name. Unfortunately there is no way for the programmer to detect and prevent this at the moment. I think this https://fpgasoftware.intel.com/?edition=lite&elq_cid=2980428&erpm_id=2702282 should have the fpga programmer.

ahopper, 2020-01-08:
@wb2qll Pete,
this link is better https://fpgasoftware.intel.com/18.1/?edition=lite&platform=windows 
you can either download the 'combined files' which has everything including stuff to write your own fpga code but is huge or in the 'additional software' tab you can download the stand alone programmer.

ahopper, 2020-01-08:
@wb2qll Pete,
btw for this sort of question you will probably get more help on the google group https://groups.google.com/forum/#!forum/hermes-lite

wb2qll, 2020-01-08:
There is nothing on this page that claims to be a stand alone programmer, and I've looked many times.Also, even tho it has my name and password account set, I can't sign in, says there's a "network error" with no further explanation.-----Original Message-----
From: ahopper 
Sent: Jan 8, 2020 1:07 AM
To: softerhardware/Hermes-Lite2 
Cc: wb2qll , Mention 
Subject: Re: [softerhardware/Hermes-Lite2] Failure to exit transmit mode (#139)

@wb2qll Pete,
this link is better https://fpgasoftware.intel.com/18.1/?edition=lite&platform=windows
you can either download the 'combined files' which has everything including stuff to write your own fpga code but is huge or in the 'additional software' tab you can download the stand alone programmer.

wb2qll, 2020-01-08:
Also it says I must download device support but I have no idea which device support file I need, haven't seen that written up anywhere.-----Original Message-----
From: ahopper 
Sent: Jan 8, 2020 1:07 AM
To: softerhardware/Hermes-Lite2 
Cc: wb2qll , Mention 
Subject: Re: [softerhardware/Hermes-Lite2] Failure to exit transmit mode (#139)

@wb2qll Pete,
this link is better https://fpgasoftware.intel.com/18.1/?edition=lite&platform=windows
you can either download the 'combined files' which has everything including stuff to write your own fpga code but is huge or in the 'additional software' tab you can download the stand alone programmer.

ahopper, 2020-01-08:
There is a tab called 'Additional Software' if you click that you should see 'Stand alone software' and just below that 'quartus prime programmer and tools'

wb2qll, 2020-01-08:
Oh, thanks, got it, it was indeed there.Is this the file I need?hl2b5up_main.jicor this one?hl2b5up_6rx.jicHopefully my final question.-----Original Message-----
From: ahopper 
Sent: Jan 8, 2020 1:24 AM
To: softerhardware/Hermes-Lite2 
Cc: wb2qll , Mention 
Subject: Re: [softerhardware/Hermes-Lite2] Failure to exit transmit mode (#139)

There is a tab called 'Additional Software' if you click that you should see 'Stand alone software' and just below that 'quartus prime programmer and tools'

ahopper, 2020-01-08:
hl2b5up_main.jic

wb2qll, 2020-01-08:
Great, thanks, so sorry to take up your time!Goodnight.-----Original Message-----
From: ahopper 
Sent: Jan 8, 2020 1:39 AM
To: softerhardware/Hermes-Lite2 
Cc: wb2qll , Mention 
Subject: Re: [softerhardware/Hermes-Lite2] Failure to exit transmit mode (#139)

hl2b5up_main.jic

wb2qll, 2020-01-10:
I've installed gateware v68 successfully. However the PTT issue remains unfixed. The HL2 stays in transmit mode when the PTT line is opened. This happens in PowerSDR, Quisk, and SDR-Console, all current versions. 

The secret to installing the USB Blaster driver was the use of a Windows 10/64 machine and installing the USB Blaster as a Blaster II.

softerhardware, 2020-01-10:
What is the PTT line? It must not be the software PTT as that I use everyday. Are you using the tip or ring of the front jack? Are you grounding the tip or ring and then releasing, or is either the tip or ring always grounded? What modes are the software in? I need a more detailed description. A video or picture would be helpful.

wb2qll, 2020-01-10:
The PTT line is, as you told me a few days ago, the ring connector of the front jack. I am operating the hardware PTT as I do every other radio I've ever used, grounding the line for transmit and opening the connection for receive. I have not seen any other indication there was any other way of doing it. The CW key input is not connected for t/r tests.If PowerSDR is in one of the CW modes, grounding PTT works correctly and puts the HL2 into transmit. Grounding the CW (tip) initiates a signal. Ungrounding the PTT line causes the HL2 to revert to receive. This is correct.I PowerSDR is in any of the voice modes, Grounding PTT puts the HL2 into transmit and a modulated signal eventuates. When the PTT line is ungrounded the HL2 continues to transmit a modulated signal. This is incorrect. Pressing MOX takes the HL2 out of transmit and into receive. In all cases when the hardware line is grounded the MOX button lights up.What's the most convenient way to send a video to you? I can do any normal means.-----Original Message-----
From: Steve Haynal 
Sent: Jan 9, 2020 9:49 PM
To: softerhardware/Hermes-Lite2 
Cc: wb2qll , Mention 
Subject: Re: [softerhardware/Hermes-Lite2] Failure to exit transmit mode (#139)

What is the PTT line? It must not be the software PTT as that I use everyday. Are you using the tip or ring of the front jack? Are you grounding the tip or ring and then releasing, or is either the tip or ring always grounded? What modes are the software in? I need a more detailed description. A video or picture would be helpful.

softerhardware, 2020-01-12:
Hi Pete,
Thanks for additional details, but I can't duplicate your problem. I've tried with Quisk and PowerSDR. I insert a spare audio cable into the key/ptt jack, and then use a jumper cable on the other end to ground either the ring. The HL2 enters TX when the ring is grounded and immediately exits when released. There must be something else going on with your setup:

** Have you double checked that you are using gateware 68? PowerSDR will identify this as 6.8.
** Check for any damage to R75,R76,R77,R78,C71 and C72. Confirm with a voltmeter that if the ring or tip is not grounded, then the ring and tip are pulled to 3.3V by R75 and R76.
** Do you see the same behavior with Quisk or SparkSDR?
** Remove the PowreSDR database and reconfigure from scratch again. Maybe something in your setup is causing this.

73,

Steve
kf7o

wb2qll, 2020-01-12:
Hi Steve, thanks for checking. Before I forget, I have the HL2 working perfectly with JS8Call. Using bog-standard settings of VAC and com0com and JS8Call believes it’s talking to a TS2000 and that is what PowerSDR is set to.

 

As to the PTT issue, yes PowerSDR and other programs identify the gateware as 68. I’m on a deadline for work this weekend so I don’t have time to check the resistors and caps. You know the print better than I but I doubt its hardware related because the PTT operation works fine when the software is in cw modes as I’ve mentioned.

 

PowerSDR shows the same issue with two different computers, one Win 7/32, the other Win 10/64 so doubt it’s a database issue. I get the same problem showing up with SDRConsole and Quisk. I haven’t been able to get Spark working properly, doesn’t accept audio input from the Delta 44 sound card and it doesn’t have a cw mode, nor does it respond to the PTT line at all. The software PTT works.

 

Voltage on the PTT line is 3.2 whether the HL2 is in receive or transmit.

 

My guess is that once the software sees the PTT line is grounded, the software goes into transmit mode, as it should. But there is a failure of the HL2 to communicate to the software that the PTT is ungrounded. Thus the software keeps the HL2 in transmit when the software in the voice modes.

 

All I can do for now.

 

73,

Pete

softerhardware, 2020-01-14:
Hi Pete,

This is a tough one as I can't fix a bug I can't replicate. We have to figure out what the difference between your setup and my setup is. Here are a few more ideas:

** Just to eliminate something different with your switch, can you plug in an audio extension cable to the front jack and use tinfoil to either engage PTT or PTT and CW. This is what I am doing and it has always been coming out of TX.

** You can do a wireshark capture of when the PTT should be released. This will tell us who (software or HL2) is holding the transmit.

** A video of the problem can often be very helpful as it will include things we are just not thinking of right now. Just a video of your PC screen with a phone should be fine. You can post it on YouTube or e-mail it to hermeslite@gmail.com.

** I am doing final testing of gateware 69. Please try that when it is released in a few days to see if there is any difference.

** On the list, you believed for a short time that gateware 68 fixed the problem. Did gateware 68 make any difference? Is the problem intermittent so that you thought it was fixed?

73,

Steve
kf7o

wb2qll, 2020-01-14:
OK Steve I can try. I brought the HL2 to show and tell at my club meeting last night, people were really interested. May be a few days 'til I can power it up again but I can tell you now:I've never used a switch for PTT, I'm just using an audio cable and touching one end to ground to put it into transmit.I would be happy to run Wireshark, but since I've never used it and never heard of it prior to your mentioning it, that might take a while to install and figure out. I did find the web page.I think I believed 68 made a difference because I had PowerSDR in cw mode where the PTT works properly. This may have changed between v67 and 68. It's my recollection the PTT did not come out of xmit when PowerSDR was in cw mode with v67.The issue is most definitely not intermittent.I'll get you a video when I next power it up.Thanks!Pete-----Original Message-----
From: Steve Haynal 
Sent: Jan 14, 2020 1:23 AM
To: softerhardware/Hermes-Lite2 
Cc: wb2qll , Mention 
Subject: Re: [softerhardware/Hermes-Lite2] Failure to exit transmit mode (#139)

Hi Pete,
This is a tough one as I can't fix a bug I can't replicate. We have to figure out what the difference between your setup and my setup is. Here are a few more ideas:
** Just to eliminate something different with your switch, can you plug in an audio extension cable to the front jack and use tinfoil to either engage PTT or PTT and CW. This is what I am doing and it has always been coming out of TX.
** You can do a wireshark capture of when the PTT should be released. This will tell us who (software or HL2) is holding the transmit.
** A video of the problem can often be very helpful as it will include things we are just not thinking of right now. Just a video of your PC screen with a phone should be fine. You can post it on YouTube or e-mail it to hermeslite@gmail.com.
** I am doing final testing of gateware 69. Please try that when it is released in a few days to see if there is any difference.
** On the list, you believed for a short time that gateware 68 fixed the problem. Did gateware 68 make any difference? Is the problem intermittent so that you thought it was fixed?
73,
Steve
kf7o

softerhardware, 2020-01-19:
This was resolved in private e-mails. An upgrade to the latest version 68 solved the issue.

closed 2020-01-19 by softerhardware