softerhardware/Hermes-Lite2 #159: After changing to the cicrx.rbf gateware gain stays on 100 rbfs for all and any receiver

waterwin, 2021-07-04:
I uploaded the alternative gateware for the 10 RX slices with SparkSDR and Quisk.
Both give the same result when trying to run a receiver in SparkSDR.
On all frequencies, in all bands the same spectrum and white noise only.
Gain stays on 100 rbfs in MAX and OFF modes.

Roll back to main.rbf works and back to 4 RX slices is working.

What else can I try to get to a working 10 RX slice Hermes Lite V2 ?

softerhardware, 2021-07-05:
Hi,

Can we move this to the Google group?
hermeslite.com

Are you running with RX bandwidth of 192kHz? To fit 10 receivers, we have
to simplify things and run at 192kHz with only CIC filtering, so only ~40%
of the bandwidth is usable.

73,

Steve
kf7o

waterwin, 2021-07-05:
Thanks for your quick reaction Steve,
I tried all bandwidth possibilities.
The issue is there when starting 1 receiver, or 2, or 3 ..... or 10
I'll copy the info to the group and do a second try today with Quisk as well

softerhardware, 2021-07-22:
I have used this 10 CIC gateware just today with no gain issues. 
https://github.com/softerhardware/Hermes-Lite2/tree/master/gateware/bitfiles/stable/20201212_72p8/variants/hl2b5up_cicrx

I think this has been solved on the Google Group so am closing the issue.

closed 2021-07-22 by softerhardware